openrocket/openrocket #2775: Warn if imported Rocksim file has nose cone "Base length" parameter non-zero

neilweinstock, 2025-04-06:
See discussion on TRF starting here: https://www.rocketryforum.com/threads/potential-cp-problem-in-rocksim.191840/post-2727827

Rocksim has a "base length" parameter that OR does not currently support.  If we see it set non-zero on an imported file, we should issue a warning because it will materially change the imported design.

JoePfeiffer, 2025-04-08:
Would it make sense to give the warning, and extend the length of the nose cone to account for the base length?

neilweinstock, 2025-04-08:
Hmm, interesting idea.  It seems like it has to be preferable than simply omitting the base extension.

I think I am in favor of this as long as the warning message explains what we're doing.

SiboVG, 2025-04-16:
We could also add a straight transition aft of the nose cone to represent the base length? The nose cone would have no shoulder; the base length transition would have the aft shoulder from the RockSim nose cone.

neilweinstock, 2025-04-16:
That's even better, two thumbs up.  Not even sure if we'd need a warning for that (but probably still a good idea).

JoePfeiffer, 2025-04-17:
Though long term it would be good to support the cylindrical section as a nose cone parameter.

closed 2026-10-25 by SiboVG