openrocket/openrocket #2789: [Enhancement] Improve UI for low-resolution screens

SiboVG, 2025-04-16:
See #2788. We should:

- Try to optimize window sizes
- Make dialogs scrollable when resized

Kruzk02, 2025-11-01:
Hi can I work on this?

SiboVG, 2025-11-01:
> Hi can I work on this?

Hi, sure, thanks! I will, however, first push some near-finished code I made a while ago for minimizing the motor chooser dialog. I'll do that in the next hour. After that, you can work on the other dialogs.

Kruzk02, 2025-11-01:
Should I keep most simple dialogs in **dialogs** package as they are, or still add a scroll pane ? I'm currently working on the **configdialog** package.

SiboVG, 2025-11-01:
> Should I keep most simple dialogs in **dialogs** package as they are, or still add a scroll pane ? I'm currently working on the **configdialog** package.

You should remove unnecessary blank space in the dialogs, but don't overdo it, and then make sure that when the user resizes the dialog to smaller than the dialog content, that the content becomes scrollable.

Kruzk02, 2025-11-02:
@SiboVG I saw Simulation dialogs expand their content when stretched, but Component configs dialog stays fixed. Do I need stretched content panel or keep the current?