openrocket/openrocket #2791: [Bug] 3D views not working (Finished, Unfinished and Figure)

dynamixbot, 2025-04-19:
### Describe the bug

I have been using OpenRocket for a long while now, but then took a break and started again. But the 3D views are not working, only the diagrams. I thought the reason was that my OpenRocket version was not updated, but after updating to 23.09 from 22.02, no luck. Reverting back to 22.02 doesn't work, and I tried deleting and re-installing OpenRocket 4 times, still no views working. I then thought that my ork files might be the issue, but opening the example rocket files and trying to view them in 3D also didn't work.

### To Reproduce

I don't know how to reproduce the issue, OpenRocket works fine on other machines but not on my laptop.

### Screenshots / .ork file

#### Screenshot(s):
*(drag-and-drop the screenshot(s) here)*

[image]
what appears when i try to see the 3D view.

[image]
screenshot of the ork when openrocket used to work - dated 2023-07-17 - note this is a different ork file than the above but it is just a duplicate of it

#### .ork file:
*(drag-and-drop the file here as a .zip file)*

[test for school.zip](https://github.com/user-attachments/files/19820059/test.for.school.zip)
one of my ork files where 3d doesn't work

[Boost-X110-(Ryzer)-v2.5.zip](https://github.com/user-attachments/files/19820063/Boost-X110-.Ryzer.-v2.5.zip)
the latest ork file of the one in the screenshots, the duplicate of the old screenshot ork.

### OpenRocket version

22.02, 23.09

### What platform are you running on?

Windows

### Additional context

Specifications of my machine:

CPU: AMD Ryzen 5 5600H with Radeon Graphics
GPU: NVIDIA GeForce GTX 1650M
RAM: 16GB DDR4 SODIMM
Storage: 256GB SSD + 1TB HDD

dynamixbot, 2025-04-20:
I think this might be of relevance, but my laptop is a Lenovo Ideapad Gaming 3 15ACH6.

dynamixbot, 2025-04-26:
> We've made a few changes in 24.12.RC.01 that seem to have helped users having this problem. Please give it a try and report back. https://github.com/openrocket/openrocket/releases/tag/release-24.12.RC.01

is there a windows x64 version?

JoePfeiffer, 2025-04-26:
> > We've made a few changes in 24.12.RC.01 that seem to have helped users having this problem. Please give it a try and report back. https://github.com/openrocket/openrocket/releases/tag/release-24.12.RC.01
> 
> is there a windows x64 version? and one more thing, which version of java does 23.09 or 24.12.RC.01 require?
https://github.com/openrocket/openrocket/releases/download/release-24.12.RC.01/OpenRocket-24.12.RC.01-installer-Windows-x86_64.exe

The installer includes the right Java runtime (which is Java 17).

dynamixbot, 2025-04-27:
> https://github.com/openrocket/openrocket/releases/download/release-24.12.RC.01/OpenRocket-24.12.RC.01-installer-Windows-x86_64.exe
>
> The installer includes the right Java runtime (which is Java 17).

i have jdk 21 lts installed, it won't override anything right?

dynamixbot, 2025-04-27:
still doesn't work, JDK 21 LTS installed, using OpenRocket 24.12.RC.01 and still no luck. still it shows the same black screen.

JoePfeiffer, 2025-04-27:
> i have jdk 21 lts installed, it won't override anything right?

Correct. It'll use 17 for OR, but not anything else.

> still doesn't work, JDK 21 LTS installed, using OpenRocket 24.12.RC.01 and still no luck. still it shows the same black screen.

At this point I'm out of ideas. Hopefully one of our Windows users (I'm on Linux) will have some more suggestions.

dynamixbot, 2025-05-01:
uhh i started using kicad today and the same issues have been popping up. could this be some sort of file preview bug in my computer or something? i can't view 3d, PCB layouts, schematic files, anything...

JoePfeiffer, 2025-05-01:
> uhh i started using kicad today and the same issues have been popping up. could this be some sort of file preview bug in my computer or something? i can't view 3d, PCB layouts, schematic files, anything...

There appears to be some sort of bug in Windows's opengl driver relating to some graphics cards, affecting a small number of users. This is the first I've heard of it affecting an application written in something other than Java, however (kicad is written in... checking... C++).

Unfortunately, none of the developers are affected by it, so we're sort of working blindly trying to fix it. Sorry...

dynamixbot, 2025-05-15:
> There appears to be some sort of bug in Windows's opengl driver relating to some graphics cards, affecting a small number of users. This is the first I've heard of it affecting an application [...]

could I get a CVE number of that bug?

JoePfeiffer, 2025-05-15:
We don't know if there is one (hence "appears to be some sort").  We've just observed many users having no problem, and a few getting bitten.

dynamixbot, 2025-05-16:
> We don't know if there is one (hence "appears to be some sort"). We've just observed many users having no problem, and a few getting bitten.

oh ok thanks

neilweinstock, 2025-05-29:
If this is affecting multiple applications should this stay open as an OR bug?

dynamixbot, 2025-05-30:
> If this is affecting multiple applications should this stay open as an OR bug?

i mean it still affects openrocket with the side effect/downside/benefit of it affecting other apps.

dynamixbot, 2025-07-11:
Issue resolved, I had to just update my laptop and 24.12.RC.01 works, not 23.09 though.

neilweinstock, 2025-07-11:
Thanks for following up.