openrocket/openrocket #2792: [Bug] The apogee keeps showing one same number

NG-Corp, 2025-04-25:
### Describe the bug

The apogee shows 786 no matter what.

### To Reproduce

1. 
2. 
3. 
...

### Screenshots / .ork file

#### Screenshot(s):
*(drag-and-drop the screenshot(s) here)*
[image]

#### .ork file:
*(drag-and-drop the file here as a .zip file)*

### OpenRocket version

Latest unstable version

### What platform are you running on?

Windows

JoePfeiffer, 2025-04-26:
We'll need more information before we can look in to this. When you say "no matter what", what different things have you done that continues to show the same apogee? Also, could you post your .ork file? Due to limitations of github, you'll need to zip it before you can upload it.

senft-research, 2025-05-07:
@NG-Corp  are you able to provide more information?

senft-research, 2025-05-19:
@JoePfeiffer what is our policy on bugs where the user does not respond after several weeks? (Just doing my bit to try and reduce the number of bug issues, and if we can't successfully replicate this bug...)

JoePfeiffer, 2025-05-19:
No further info from user.

closed 2025-05-19 by JoePfeiffer