openrocket/openrocket #2794: java.lang.IllegalArgumentException: Cannot project a degenerate polygon onto a 2D plane

emceeaich, 2025-05-01:
---------- Bug report ----------

Please include a description about what actions you were performing when the exception occurred:
(You can edit text directly in this window)

1. Saving a nose cone as an obj file
2.
3.

If possible, please send us the .ork file that caused the bug.

Include your email address (optional; it helps if we can contact you in case we need additional information):

(Do not modify anything below this line.)
---------- Exception stack trace ----------
java.lang.IllegalArgumentException: Cannot project a degenerate polygon onto a 2D plane
at info.openrocket.core.file.wavefrontobj.TriangulationHelper.projectVertexOntoXYPlane(TriangulationHelper.java:173)
at info.openrocket.core.file.wavefrontobj.TriangulationHelper.createProjectedPolygon(TriangulationHelper.java:144)
at info.openrocket.core.file.wavefrontobj.TriangulationHelper.generateCDTFaces(TriangulationHelper.java:76)
at info.openrocket.core.file.wavefrontobj.TriangulationHelper.constrainedDelaunayTriangulate(TriangulationHelper.java:48)
at info.openrocket.core.file.wavefrontobj.export.OBJExporterFactory.doExport(OBJExporterFactory.java:180)
at info.openrocket.swing.gui.main.BasicFrame.saveWavefrontOBJFile(BasicFrame.java:1728)
at info.openrocket.swing.gui.main.BasicFrame.saveAsWavefrontOBJ(BasicFrame.java:1715)
at info.openrocket.swing.gui.main.BasicFrame.exportWavefrontOBJAction(BasicFrame.java:1708)
at info.openrocket.swing.gui.main.RocketActions$ExportOBJAction.actionPerformed(RocketActions.java:1274)
at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1972)
at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2313)
at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405)
at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262)
at java.desktop/javax.swing.AbstractButton.doClick(AbstractButton.java:374)
at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:1028)
at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:1072)
at java.desktop/java.awt.AWTEventMulticaster.mouseReleased(AWTEventMulticaster.java:297)
at java.desktop/java.awt.Component.processMouseEvent(Component.java:6626)
at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3389)
at java.desktop/java.awt.Component.processEvent(Component.java:6391)
at java.desktop/java.awt.Container.processEvent(Container.java:2266)
at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5001)
at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2324)
at java.desktop/java.awt.Component.dispatchEvent(Component.java:4833)
at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4948)
at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4575)
at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4516)
at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2310)
at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2780)
at java.desktop/java.awt.Component.dispatchEvent(Component.java:4833)
at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:775)
at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:720)
at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:714)
at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:97)
at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:747)
at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745)
at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:744)
at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)

---------- Thread information ----------
Thread[AWT-EventQueue-0,6,main]

---------- System information ----------
OpenRocket version: 24.12.RC.01
OpenRocket source: default
[... 404 lines cut ...]
987863 198896.571 STDERR (SwingExceptionHandler.java:50) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2310)
987864 198896.571 STDERR (SwingExceptionHandler.java:50) at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2780)
987865 198896.571 STDERR (SwingExceptionHandler.java:50) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4833)
987866 198896.571 STDERR (SwingExceptionHandler.java:50) at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:775)
987867 198896.571 STDERR (SwingExceptionHandler.java:50) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:720)
987868 198896.571 STDERR (SwingExceptionHandler.java:50) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:714)
987869 198896.571 STDERR (SwingExceptionHandler.java:50) at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
987870 198896.571 STDERR (SwingExceptionHandler.java:50) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
987871 198896.571 STDERR (SwingExceptionHandler.java:50) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:97)
987872 198896.571 STDERR (SwingExceptionHandler.java:50) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:747)
987873 198896.571 STDERR (SwingExceptionHandler.java:50) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745)
987874 198896.571 STDERR (SwingExceptionHandler.java:50) at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
987875 198896.571 STDERR (SwingExceptionHandler.java:50) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
987876 198896.571 STDERR (SwingExceptionHandler.java:50) at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:744)
987877 198896.571 STDERR (SwingExceptionHandler.java:50) at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
987878 198896.571 STDERR (SwingExceptionHandler.java:50) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
987879 198896.571 STDERR (SwingExceptionHandler.java:50) at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
987880 198896.571 STDERR (SwingExceptionHandler.java:50) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
987881 198896.571 STDERR (SwingExceptionHandler.java:50) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
987882 198896.571 STDERR (SwingExceptionHandler.java:50) at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
987883 198896.571 INFO (SwingExceptionHandler.java:62) Exception handler running on EDT, showing dialog
987884 198896.572 INFO (SwingExceptionHandler.java:202) Showing Exception dialog
987885 198901.892 USER (SwingExceptionHandler.java:221) User requested sending bug report
987886 198901.902 DEBUG (BugReportDialog.java:230) Found jar file using codeSource
---------- End of bug report ----------

SiboVG, 2025-05-03:
Hi, could you please send us the .ork file with the nose cone that causes this bug? You can ZIP the file and upload it here as a comment on GitHub. Thanks.

emceeaich, 2025-05-03:
Here's the file [punisher-nano.ork.zip](https://github.com/user-attachments/files/20020516/punisher-nano.ork.zip)

One thing I found: I changed the nose cone from an VK to an ogive. I was able to save it as an obj. Then if I changed it back to an ogive, I was able to save it. I hope that helps. The saved file is the version with an ogive.

SiboVG, 2025-05-06:
> Here's the file [punisher-nano.ork.zip](https://github.com/user-attachments/files/20020516/punisher-nano.ork.zip)
>
> One thing I found: I changed the nose cone from an VK to an ogive. I was able to [...]

Thanks! I do have difficulty replicating the issue. Just to be sure: the .ork file you sent is with an ogive nose cone, which normally does not give an error? Even if I set the nose cone to Haack series with shape parameter 0, I don't get an error.

Could you perhaps send the file again, saved in a state that reliably gives you the error? And could you also please share the export settings you use in the .obj export dialog, including the advanced settings (click "Show Advanced Options" in the export dialog?

[image]

PR #2890 (2025-09-20): Account for zero-thickness and zero-length components in OBJ export
This PR fixes #2881, fixes #2794, and fixes #2611. OBJ exporting now better accounts for components that have a 0 thickness or length.