openrocket/openrocket #2800: [Bug] Unable to open file

ShootingStars76, 2025-05-11:
### Describe the bug

Have been working on a goofy rocket kinda as a meme project and was working on it for several days so have closed it / opened it several times. One day I tried to open it and it refused too and threw an error. After i clicked OK openrocket would then open but no file would be loaded. 

### To Reproduce

1. Open file 
2. Throws error with long bug code 
3. Try again same thing happens 
...

### Screenshots / .ork file

#### Screenshot(s):
*(drag-and-drop the screenshot(s) here)*

[image]

#### .ork file:

*(drag-and-drop the file here as a .zip file)*
[3in to 54mm 2 stage.zip](https://github.com/user-attachments/files/20146831/3in.to.54mm.2.stage.zip)

### OpenRocket version

24.12.RC.01

### What platform are you running on?

Windows

senft-research, 2025-05-11:
This issue is related to #2777, with the root cause being identified, and is actively being worked on.

JoePfeiffer, 2025-05-11:
Thank you for your report -- this is a duplicate of #2777

ShootingStars76, 2025-05-11:
Welp I have done it again with the fixed file they gave me in the OpenRocket discord. All I changed was the booster motor and the launch site elevation (between balls and SLV). 

[image]

[rocketFIX.zip](https://github.com/user-attachments/files/20150094/rocketFIX.zip)

senft-research, 2025-05-11:
To add clarity, this is happening because you are changing your model, but then not updating your simulations. 

The issue I believe is the root cause is that the simulations that are "out of date" are saving warnings / messages / events that no longer exists / components that no longer exist (depends on exactly what the message was before update). 

The easiest way to prevent this (temporarily as the issue is resolved) is to always update / run your simulations after updating the model. 

The root cause of this issue (and the aforementioned #2777) is being worked on.

JoePfeiffer, 2025-05-11:
> To add clarity, this is happening because you are changing your model, but then not updating your simulations.
>
> The issue I believe is the root cause is that the simulations that are "out of date" [...]

This  is triggered slightly differently than #2777 -- it actually inserts a problematic event when the sim is rerun (that's why I reopened this issue).

JoePfeiffer, 2025-05-11:
> Edit: just as i posted this it looks like the issue has been band aided for now (just included a parachute) but ill leave this up so you have more to work with if needed

Yes, it's been band-aided but not fixed. Please don't close this issue.

PR #2801 (2025-05-15): Hook up SimulationStatus WarningSet to FlightData before performing sanity check on configuration.
Fixes #2800
closed 2025-05-15 by JoePfeiffer