openrocket/openrocket #2804: [Bug] Saving an .ork with 2 post-ground-hit warnings makes file unloadable with a bug warning

jimmiedave, 2025-05-16:
### Describe the bug

### Description

When a rocket motor is not powerful enough to keep the model in the air until recovery deployment, deployment happens after ground-hit, which creates two post-ground-hit warnings. Saving this file with all simulation data creates an otherwise-valid .ork which cannot be opened due to the presence of two warnings in the file that are missing something. The bug warning that appears does not present a GUID for the problematic warnings in the file.

Editing the underlying rocket.ork and removing the <warning> block and the <simwarn> sections makes the file loadable again.

### Expectation:

No matter the content, the saved file should be able to be opened.

### To Reproduce

1. Open OpenRocket 24.12 RC.01
2. File>Open example>A simple model rocket.
3. OK "Rocket configuration" dialog.
4. Click "Motors & Configuration" tab.
5. [New Configuration].
6. Add an Estes 1/4A3 motor (guaranteed to be too small for a successful flight).
7. OK the Motor selection dialog.
8. Click "Flight simulations" tab.
9. Select All.
10. [Run simulations].
11. Note that "Simulation 6" has 2 warnings.
12. File>Save as...
13. **important** Optionally fill, but definitely OK the "Save Design Info" dialog (to work around #2680).
14. In the "Save as OpenRocket ork file" dialog, ensure "All simulated data" is selected.
15. Name the file and save it.
16. If the "Save options" dialog appears, ensure that "All simulated data" is selected once more, and OK it.
17. Quit OpenRocket
18. Re-open OpenRocket and open the just-saved file, or double-click it to open OpenRocket.
19. **BUG** file will not open with "BugException: BUG: Unknown error when opening file"

### Screenshots / .ork file

[image]

[junk2.ork.zip](https://github.com/user-attachments/files/20236659/junk2.ork.zip)

### OpenRocket version

24.12 RC01

### What platform are you running on?

macOS

### Additional context

[StackTrace.txt.zip](https://github.com/user-attachments/files/20236705/StackTrace.txt.zip)

senft-research, 2025-05-19:
Could you provide the context / information about the 2 warnings you mention in your replication steps.

jimmiedave, 2025-05-19:
[image]

JoePfeiffer, 2025-05-20:
This is a bug in my code (which I thought I'd fixed...). I'll look at it.

PR #2826 (2025-07-03): Attach event causing warning to "event after landing" event when saving .ork
A sim warning event caused by an event after landing ends up having three parts:

The event triggering everything (for instance, a recovery deployment occurring after landing)
A warning about that event
A simulation warning event, caused by the warning.

Nothing was hooking the original event up to either the warning or the warning event in the .ork file; this ended up causing a null pointer exception.

With this PR, the original event and the warning are both hooked up to the sim warning event so the whole chain can  [... cut]
closed 2025-07-03 by JoePfeiffer