openrocket/openrocket #2805: [Bug] Minor: "Save as..." asks twice whether to save all simulated data the first time per file in a session

jimmiedave, 2025-05-16:
### Describe the bug

With a given OpenRocket file, in a given session of OpenRocket, the first time user executes "Save as...", she's given both the "Save as OpenRocket ork file" dialog, which features a section asking whether to save all simulated data or just summary data, **AND** the "Save options" dialog, which asks the same question.

The settings of the two dialogs do not necessarily agree, and selection of one option in the Save as... dialog does not affect the gadgets in the Save options dialog. The behavior of this process (which data are saved, how chosen) is untested, and undefined so far.

Without closing this OpenRocket file or session, a second "Save as..." action will **only** bring up the first "Save as OpenRocket ork file" dialog, and presumably honor the selection made there.

It's a weird behavior. Is it by design?

### To Reproduce

1. Open OpenRocket.
2. File>Open example>A simple model rocket.
3. OK Rocket configuration dialog.
4. File>Save As...
5. **important** at a minimum, OK the "Save design info" dialog to work around #2680.
6. Note setting of "Save options" section of "Save as OpenRocket ork file" dialog.
7. Pick a name and save.
8. Note that the "Save options" dialog appears. The options may be different than the prior dialog.
9. OK Save options.
10. File>Save As...
11. OK the "Save design info" dialog.
12. Note that the "Save options" dialog appears. The options may be different than the prior dialog.
13. Pick a new file name and Save.
14. Note that **no** Save options dialog appears.

### Screenshots / .ork file

[image]

[image]

### OpenRocket version

24.12 RC.01

### What platform are you running on?

macOS

senft-research, 2025-05-19:
@JoePfeiffer  is this intentional behaviour for some reason? If not I'd be happy to be assigned this issue.

JoePfeiffer, 2025-05-20:
I see the same behavior under Linux; no, it's not the intended behavior. Go for it!

senft-research, 2025-05-20:
Just so its written down somewhere, I've identified the problem as caused by `StorageOptionChooser.verifyStorageOptions` . It is a method that is specifically called during the `BasicFrame.saveAsOpenRocket` call. 

Does this behavior happen on Windows aswell as macOS / Linux?

senft-research, 2025-05-20:
Update: Figured out the bug and found a solution that works, but there is an underlying problem. There is no unifying interface for "choosers" used by the application. 

The issue is here in `BasicFrame.openFileSaveAsDialog`

```java
OBJOptionChooser objChooser = null;
		
		if (chooser.getAccessory() instanceof OBJOptionChooser) {
			objChooser = (OBJOptionChooser) chooser.getAccessory();
		}
		int option = chooser.showSaveDialog(BasicFrame.this);

		if (option != JFileChooser.APPROVE_OPTION) {
			log.info(Markers.USER_MARKER, "User decided not to save, option=" + option);
			return null;
		}
		// Store the OBJ options
		if (objChooser != null) {
			objChooser.storeOptions(document.getDefaultOBJOptions(), true);

			// We need to separately store the preference options, because the export children option can be
			// automatically selected based on whether only component assemblies are selected. We don't want to
			// store that state in the preferences.
			OBJExportOptions prefOptions = new OBJExportOptions(rocket);
			objChooser.storeOptions(prefOptions, false);
			prefs.saveOBJExportOptions(prefOptions);
		}
```

The code specifically checks if the accessory of the OptionChooser is an instance of `OBJOptionChooser` , but does not check for if it is a `StorageOptionChooser`. 

This highlights a fundamental scalability problem if we were to add different accessories to the save as dialog in future. So I am going to do a refactor to add some sort of interface to `OBJOptionChooser` and `StorageOptionChooser` respectfully, and then have a generalized call of some sort of `.store` method and also handle that custom OBJ based logic

jimmiedave, 2025-05-21:
> Just so its written down somewhere, I've identified the problem as caused by `StorageOptionChooser.verifyStorageOptions` . It is a method that is specifically called during the `BasicFrame.saveAsOpenRocket` call.
> 
> Does this behavior happen on Windows aswell as macOS / Linux?

I think you've got this, but yes, reproed on Win11 ARM in Parallels VM

PR #2808 (2025-06-05): StorageOptions dialog being shown twice fix / OptionChooser Refactor
Fixing the bug where storage options dialog would pop up, regardless of save as options being selected. 

Also done refactor of 2 of the Option Chooser classes to a new interface to allow for ease of scalability in the future. 

Resolves #2805