openrocket/openrocket #2812: Error clicking "Flight Simulations

rocketrocket1-bit, 2025-05-23:
1. Opened existing file
2.Clicked Flight Simulations
3.

If possible, please send us the .ork file that caused the bug.

Include your email address (optional; it helps if we can contact you in case we need additional information):

(Do not modify anything below this line.)
---------- Exception stack trace ----------
java.lang.NullPointerException: Cannot invoke "info.openrocket.core.rocketcomponent.RocketComponent.getName()" because "sources[i]" is null at info.openrocket.core.logging.Message.addSourcesToMessageText(Message.java:47) at info.openrocket.core.logging.Message.toString(Message.java:34) at info.openrocket.swing.gui.main.SimulationPanel.getSimulationWarningsToolTip(SimulationPanel.java:858) at info.openrocket.swing.gui.main.SimulationPanel$WarningsBox.updateContent(SimulationPanel.java:1250) at info.openrocket.swing.gui.main.SimulationPanel$WarningsBox.<init>(SimulationPanel.java:1232) at info.openrocket.swing.gui.main.SimulationPanel$SimulationTableModel$2.getValueAt(SimulationPanel.java:1381) at info.openrocket.swing.gui.adaptors.ColumnTableModel.getValueAt(ColumnTableModel.java:58) at java.desktop/javax.swing.JTable.getValueAt(JTable.java:2763) at java.desktop/javax.swing.JTable.prepareRenderer(JTable.java:5795) at java.desktop/javax.swing.plaf.basic.BasicTableUI.paintCell(BasicTableUI.java:2210) at java.desktop/javax.swing.plaf.basic.BasicTableUI.paintCells(BasicTableUI.java:2112) at java.desktop/javax.swing.plaf.basic.BasicTableUI.paint(BasicTableUI.java:1908) at com.formdev.flatlaf.ui.FlatTableUI.paint(FlatTableUI.java:539) at java.desktop/javax.swing.plaf.ComponentUI.update(ComponentUI.java:161) at java.desktop/javax.swing.JComponent.paintComponent(JComponent.java:842) at java.desktop/javax.swing.JComponent.paint(JComponent.java:1119) at java.desktop/javax.swing.JComponent.paintChildren(JComponent.java:952) at java.desktop/javax.swing.JComponent.paint(JComponent.java:1128) at java.desktop/javax.swing.JViewport.paint(JViewport.java:736) at java.desktop/javax.swing.JComponent.paintChildren(JComponent.java:952) at java.desktop/javax.swing.JComponent.paint(JComponent.java:1128) at java.desktop/javax.swing.JComponent.paintChildren(JComponent.java:952) at java.desktop/javax.swing.JComponent.paint(JComponent.java:1128) at java.desktop/javax.swing.JComponent.paintChildren(JComponent.java:952) at java.desktop/javax.swing.JComponent.paint(JComponent.java:1128) at java.desktop/javax.swing.JComponent.paintChildren(JComponent.java:952) at java.desktop/javax.swing.JSplitPane.paintChildren(JSplitPane.java:1030) at java.desktop/javax.swing.JComponent.paint(JComponent.java:1128) at java.desktop/javax.swing.JComponent.paintToOffscreen(JComponent.java:5311) at java.desktop/javax.swing.RepaintManager$PaintManager.paintDoubleBufferedImpl(RepaintManager.java:1657) at java.desktop/javax.swing.RepaintManager$PaintManager.paintDoubleBuffered(RepaintManager.java:1632) at java.desktop/javax.swing.RepaintManager$PaintManager.paint(RepaintManager.java:1570) at java.desktop/javax.swing.RepaintManager.paint(RepaintManager.java:1337) at java.desktop/javax.swing.JComponent._paintImmediately(JComponent.java:5259) at java.desktop/javax.swing.JComponent.paintImmediately(JComponent.java:5069) at java.desktop/javax.swing.RepaintManager$4.run(RepaintManager.java:879) at java.desktop/javax.swing.RepaintManager$4.run(RepaintManager.java:862) at java.base/java.security.AccessController.doPrivileged(AccessController.java:399) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86) at java.desktop/javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:862) at java.desktop/javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:835) at java.desktop/javax.swing.RepaintManager.prePaintDirtyRegions(RepaintManager.java:784) at java.desktop/javax.swing.RepaintManager$ProcessingRunnable.run(RepaintManager.java:1898) at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:318) at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:773) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:720) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:714) at java.base/java.security.AccessController.doPrivileged(AccessController.java:399) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86) at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742) at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124) at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101) at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)

---------- Thread information ----------
Thread[AWT-EventQueue-0,6,main]

---------- System information ----------
OpenRocket version: 24.12.beta.01
OpenRocket source: default
OpenRocket location: C:\Program Files\OpenRocket\.\jar\OpenRocket.jar
User-defined thrust curves location: C:\Users\Parents fast desktop\AppData\Roaming\OpenRocket\ThrustCurves
LAF: com.formdev.flatlaf.FlatLightLaf
JOGL version: null
Current default locale: en_US
System properties:
apple.awt.application.appearance=system
exe4j.consoleCodepage=cp0
exe4j.isInstall4j=true
exe4j.launchName=C:\Program Files\OpenRocket\OpenRocket.exe
exe4j.moduleName=C:\Program Files\OpenRocket\OpenRocket.exe
exe4j.semaphoreName=Local\c:_program_files_openrocket_openrocket.exe0
exe4j.tempDir=
exe4j.unextractedPosition=0
file.encoding=Cp1252
file.separator=\
flatlaf.uiScale=1
flatlaf.uiScale.enabled=true
install4j.appDir=C:\Program Files\OpenRocket\
install4j.exeDir=C:\Program Files\OpenRocket\
install4j.launcherId=59
install4j.swt=false
java.class.path=C:\Program Files\OpenRocket\.install4j\i4jruntime.jar;C:\Program Files\OpenRocket\.\jar\OpenRocket.jar
java.class.version=61.0
java.home=c:\program files\openrocket\jre
java.io.tmpdir=C:\Users\PARENT~1\AppData\Local\Temp\
java.library.path=c:\program files\openrocket\jre\bin;C:\WinAVR-20100110\bin;C:\WinAVR-20100110\utils\bin;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\ProgramData\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Common Files\Adobe\AGL;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\PuTTY\;C:\Users\Parents fast desktop\AppData\Local\Microsoft\WindowsApps;
java.runtime.name=OpenJDK Runtime Environment
java.runtime.version=17.0.13+12-LTS
java.specification.maintenance.version=1
java.specification.name=Java Platform API Specification
java.specification.vendor=Oracle Corporation
java.specification.version=17
java.util.Arrays.useLegacyMergeSort=true
java.vendor=BellSoft
java.vendor.url=https://bell-sw.com/
java.vendor.url.bug=https://bell-sw.com/support
java.version=17.0.13
java.version.date=2024-10-15
java.vm.compressedOopsMode=Zero based
java.vm.info=mixed mode
java.vm.name=OpenJDK 64-Bit Server VM
java.vm.specification.name=Java Virtual Machine Specification
java.vm.specification.vendor=Oracle Corporation
java.vm.specification.version=17
java.vm.vendor=BellSoft
java.vm.version=17.0.13+12-LTS
jdk.debug=release
jnlp.jogamp.tmp.cache.root=jln8912504688938745793
line.separator=\u000d\u000a
native.encoding=Cp1252
os.arch=amd64
[... 196 lines cut ...]
2169 24.355 STDERR (SwingExceptionHandler.java:50) at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
2170 24.355 STDERR (SwingExceptionHandler.java:50) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
2171 24.355 STDERR (SwingExceptionHandler.java:50) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:117)
2172 24.355 STDERR (SwingExceptionHandler.java:50) at java.desktop/java.awt.WaitDispatchSupport$2.run(WaitDispatchSupport.java:191)
[... cut ...]
2188 24.355 STDERR (SwingExceptionHandler.java:50) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
2189 24.355 STDERR (SwingExceptionHandler.java:50) at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
2190 24.355 WARN (SwingExceptionHandler.java:53) Exception is currently being handled, ignoring
2191 38.458 USER (SwingExceptionHandler.java:221) User requested sending bug report
2192 38.463 DEBUG (BugReportDialog.java:228) Found jar file using codeSource
---------- End of bug report ----------[](url)

neilweinstock, 2025-05-29:
@rocketrocket1-bit can you please zip the ORK file and attach it here?

SiboVG, 2025-07-14:
Duplicate of #2777.

closed 2025-07-14 by SiboVG