openrocket/openrocket #2820: [Enhancement] Don't allow all stages to be disabled

SiboVG, 2025-06-19:
Currently, you can disable all stages, leaving you with an empty design. We shouldn't allow this to happen. There is no good reason to do so, and it confuses people (we just had a discussion on Discord that a user experienced a "bug" where his rocket display disappeared).

neilweinstock, 2025-06-20:
Might it also make sense to always open with all stages visible?  I'm not sure stage visibility needs to be saved in the file, much in the same way warnings are always displayed when the file is first opened, regardless of whether they were shown when the file was saved.

SiboVG, 2025-06-22:
> Might it also make sense to always open with all stages visible? I'm not sure stage visibility needs to be saved in the file, much in the same way warnings are always displayed when the file is first opened, regardless of whether they were shown when the file was saved.

I wouldn't do that. We should do this if all stages are disabled in the file, but otherwise I would retain stage activeness in the file. Some users may actively use this feature and I'm sure they wouldn't want to revert their settings every time they open the design.

PR #2844 (2025-07-15): [#2820] Always force one stage to be enabled + better visuals for disabled stage
This PR fixes #2820. When there is only one active stage left, it cannot be disabled. I also added an icon to the stage selector button when the stage is disabled, to make it more visually distinct. When a button is disabled, its text is in italics.

https://github.com/user-attachments/assets/ac7ff201-819c-4942-9bfb-71a843d969e9