openrocket/openrocket #2843: [Bug] Linux build succeeds but sticks at splash screen when launched with java -jar, fails while loading motors when using gradlew run

questionablerocket, 2025-07-15:
### Describe the bug

---------- Bug report ----------
OS is Debian, with identical problem with both bullseye and bookworm releases.

Build steps are:

git pull https://github.com/openrocket/openrocket.git
git submodule init
git submodule update --remote
git fetch && git pull
gradlew clean submoduleUpdate build dist

Build succeeds, but when the resultant java file is called with java -jar <jar location>, it gets stuck on the splash screen.  Launching with gradlew run, I can see it appear to stick when it's loading motors, and I get the below stack trace (which I have edited only to protect username).  There is no ork involved; this happens simply launching the program.

The above build steps have worked for me since the transition from ant to gradle until sometime after April 22nd, the date of my last successful build on Linux.  These steps continue to succeed on MacOS.

(Do not modify anything below this line.)
---------- Exception stack trace ----------
java.lang.NullPointerException: Cannot invoke "info.openrocket.core.file.iterator.FileIterator.hasNext()" because "iterator" is null
at info.openrocket.core.database.MotorDatabaseLoader.loadSerializedMotorDatabase(MotorDatabaseLoader.java:91)
at info.openrocket.core.database.MotorDatabaseLoader.loadDatabase(MotorDatabaseLoader.java:58)
at info.openrocket.core.database.AsynchronousDatabaseLoader.doLoad(AsynchronousDatabaseLoader.java:108)
at info.openrocket.core.database.AsynchronousDatabaseLoader$LoadingThread.run(AsynchronousDatabaseLoader.java:144)

---------- Thread information ----------
Thread[DatabaseLoadingThread,1,main]

---------- System information ----------
OpenRocket version: 24.12.RC.01
OpenRocket source: default
OpenRocket location: /home/username/git/openrocket/core/build/libs/core-24.12.RC.01.jar
User-defined thrust curves location: /home/username/.openrocket/ThrustCurves
LAF: com.formdev.flatlaf.FlatLightLaf
JOGL version: 2.5.0
Current default locale: en_US
JVM Arguments: -Dfile.encoding=UTF-8 -Duser.country=US -Duser.language=en -Duser.variant
System properties:
file.encoding=UTF-8
file.separator=/
flatlaf.uiScale=1.0
java.class.path=/home/username/git/openrocket/build/libs/OpenRocket-24.12.RC.01.jar:/home/username/git/openrocket/swing/build/libs/swing-24.12.RC.01.jar:/home/username/git/openrocket/core/build/libs/core-24.12.RC.01.jar:/home/username/git/openrocket/core/libs/script-api-1.0.jar:/home/username/.gradle/caches/modules-2/files-2.1/org.glassfish.jaxb/jaxb-runtime/4.0.5/ca84c2a7169b5293e232b9d00d1e4e36d4c3914a/jaxb-runtime-4.0.5.jar:/home/username/.gradle/caches/modules-2/files-2.1/org.glassfish.jaxb/jaxb-core/4.0.5/7b4b11ea5542eea4ad55e1080b23be436795b3/jaxb-core-4.0.5.jar:/home/username/.gradle/caches/modules-2/files-2.1/jakarta.xml.bind/jakarta.xml.bind-api/4.0.2/6cd5a999b834b63238005b7144136379dc36cad2/jakarta.xml.bind-api-4.0.2.jar:/home/username/.gradle/caches/modules-2/files-2.1/org.commonmark/commonmark/0.21.0/c98f0473b17c87fe4fa2fc62a7c6523a2fe018f0/commonmark-0.21.0.jar:/home/username/.gradle/caches/modules-2/files-2.1/ch.qos.logback/logback-classic/1.5.0/d1d041eb97fe363b868ea6794ec93e6c96f83d29/logback-classic-1.5.0.jar:/home/username/.gradle/caches/modules-2/files-2.1/com.github.Dansoftowner/jSystemThemeDetector/3.8/40c89a2daa38cc277db8495f044b695b91804f4e/jSystemThemeDetector-3.8.jar:/home/username/.gradle/caches/modules-2/files-2.1/com.github.oshi/oshi-core/5.8.6/2af4a736c1a80bda7b558a48218fb3b0195282e1/oshi-core-5.8.6.jar:/home/username/.gradle/caches/modules-2/files-2.1/org.slf4j/slf4j-api/2.0.12/48f109a2a6d8f446c794f3e3fa0d86df0cdfa312/slf4j-api-2.0.12.jar:/home/username/.gradle/caches/modules-2/files-2.1/com.opencsv/opencsv/5.7.1/d707c095bc8c7c22fb3e377de774458a76229da4/opencsv-5.7.1.jar:/home/username/.gradle/caches/modules-2/files-2.1/org.apache.commons/commons-text/1.10.0/3363381aef8cef2dbc1023b3e3a9433b08b64e01/commons-text-1.10.0.jar:/home/username/.gradle/caches/modules-2/files-2.1/org.apache.commons/commons-lang3/3.12.0/c6842c86792ff03b9f1d1fe2aab8dc23aa6c6f0e/commons-lang3-3.12.0.jar:/home/username/.gradle/caches/modules-2/files-2.1/com.google.inject/guice/7.0.0/ccc518677b9367d45f99dfa592c9b039f07687d3/guice-7.0.0.jar:/home/username/.gradle/caches/modules-2/files-2.1/org.graalvm.js/js-scriptengine/23.0.3/5068494d25fbdaa790e464dd94c58a42521a25a/js-scriptengine-23.0.3.jar:/home/username/.gradle/caches/modules-2/files-2.1/org.graalvm.js/js/23.0.3/5779e51a90a98f007d470bb49d8b6e8c55480ca8/js-23.0.3.jar:/home/username/.gradle/caches/modules-2/files-2.1/org.graalvm.regex/regex/23.0.3/6148a2296380b60c68b27bdde0ca065d78e6d965/regex-23.0.3.jar:/home/username/.gradle/caches/modules-2/files-2.1/org.graalvm.truffle/truffle-api/23.0.3/c746147ab42b980fb09a9241bb3ec5ff822e9841/truffle-api-23.0.3.jar:/home/username/.gradle/caches/modules-2/files-2.1/org.graalvm.sdk/graal-sdk/23.0.3/217d0fe550b86feb3f0528e46d99856c587c3906/graal-sdk-23.0.3.jar:/home/username/.gradle/caches/modules-2/files-2.1/com.ibm.icu/icu4j/72.1/bc9057df4b5efddf7f6d1880bf7f3399f4ce5633/icu4j-72.1.jar:/home/username/.gradle/caches/modules-2/files-2.1/io.github.classgraph/classgraph/4.8.165/d7237a1fc235030b7b548eb3d671f714da01e50b/classgraph-4.8.165.jar:/home/username/.gradle/caches/modules-2/files-2.1/org.locationtech.jts/jts-core/1.19.0/3ff3baa0074445384f9e0068df81fbd0a168395a/jts-core-1.19.0.jar:/home/username/.gradle/caches/modules-2/files-2.1/de.javagl/obj/0.4.0/551dfcd4074cd8c9382941fe4f9d5b30f47d8460/obj-0.4.0.jar:/home/username/.gradle/caches/modules-2/files-2.1/org.ow2.asm/asm-commons/9.7.1/406c6a2225cfe1819f102a161e54cc16a5c24f75/asm-commons-9.7.1.jar:/home/username/.gradle/caches/modules-2/files-2.1/org.ow2.asm/asm-tree/9.7.1/3a53139787663b139de76b627fca0084ab60d32c/asm-tree-9.7.1.jar:/home/username/.gradle/caches/modules-2/files-2.1/org.ow2.asm/asm/9.7.1/f0ed132a49244b042cd0e15702ab9f2ce3cc8436/asm-9.7.1.jar:/home/username/.gradle/caches/modules-2/files-2.1/org.eclipse.angus/angus-activation/2.0.2/41f1e0ddd157c856926ed149ab837d110955a9fc/angus-activation-2.0.2.jar:/home/username/.gradle/caches/modules-2/files-2.1/jakarta.activation/jakarta.activation-api/2.1.3/fa165bd70cda600368eee31555222776a46b881f/jakarta.activation-api-2.1.3.jar:/home/username/.gradle/caches/modules-2/files-2.1/org.eclipse/yasson/2.0.1/5900313b20bbe395a386369b05d847f1feb488e3/yasson-2.0.1.jar:/home/username/.gradle/caches/modules-2/files-2.1/org.glassfish/jakarta.json/2.0.1/da6c75e591cf2a89ddc479cbdbd8fe5f41d3efc9/jakarta.json-2.0.1.jar:/home/username/.gradle/caches/modules-2/files-2.1/org.glassfish/jakarta.json/2.0.1/459368d8c32651ac61d9c9ce4cf2a3b14f4b73cc/jakarta.json-2.0.1-module.jar:/home/username/.gradle/caches/modules-2/files-2.1/com.itextpdf/itextpdf/5.5.13.3/5282897d446c3b736d63ef6e83043eb6f1943b4e/itextpdf-5.5.13.3.jar:/home/username/.gradle/caches/modules-2/files-2.1/org.jfree/jcommon/1.0.24/d5016f7703896c38f9e511f8a9edda2585cbac8b/jcommon-1.0.24.jar:/home/username/.gradle/caches/modules-2/files-2.1/org.jfree/jfreechart/1.5.4/9a5edddb05a3ca4fbc0628c594e6641a6f36a3b4/jfreechart-1.5.4.jar:/home/username/.gradle/caches/modules-2/files-2.1/com.miglayout/miglayout-swing/11.0/ea244be3c4a16c541413c4febdee539b348c744b/miglayout-swing-11.0.jar:/home/username/.gradle/caches/modules-2/files-2.1/com.miglayout/miglayout-core/11.0/99ecf243c6a64a038a568dbf8421928db9b5c3b2/miglayout-core-11.0.jar:/home/username/.gradle/caches/modules-2/files-2.1/com.fifesoft/rsyntaxtextarea/3.4.0/53c18ed94af899997e469a859b89ba3bd4cf51c7/rsyntaxtextarea-3.4.0.jar:/home/username/.gradle/caches/modules-2/files-2.1/io.github.g00fy2/versioncompare/1.5.0/ac76ded4ff83501ff5811bfb1a3ef708a2f08a7b/versioncompare-1.5.0.jar:/home/username/.gradle/caches/modules-2/files-2.1/com.formdev/flatlaf-extras/3.6/323b17b2635057624962344304a81f544f9c1663/flatlaf-extras-3.6.jar:/home/username/.gradle/caches/modules-2/files-2.1/com.formdev/flatlaf-intellij-themes/3.6/21e9dfbdfd7b0e09b43ac9e375076e4cb15fe28d/flatlaf-intellij-themes-3.6.jar:/home/username/.gradle/caches/modules-2/files-2.1/com.formdev/flatlaf/3.6/49f388b57537c286b466dee84a6d01ba57a59e9/flatlaf-3.6.jar:/home/username/.gradle/caches/modules-2/files-2.1/ch.qos.logback/logback-core/1.5.0/ac245eb4d95f57bbda4d3fff71a707ecfedfa3ef/logback-core-1.5.0.jar:/home/username/.gradle/caches/modules-2/files-2.1/org.jogamp.gluegen/gluegen-rt-main/2.5.0/6a86a9ce357c085e3052efcf639447f9957b82b8/gluegen-rt-main-2.5.0.jar:/home/username/.gradle/caches/modules-2/files-2.1/org.jogamp.jogl/jogl-all-main/2 [... cut ...]
1. Build using above steps on Debian bookworm or bullsye with openjdk 17.  A VirtualBox VM running bullsye is able to reproduce this problem.
2. Run using java -jar or gradlew run
3. 
...

### Screenshots / .ork file

#### Screenshot(s):
*(drag-and-drop the screenshot(s) here)*

#### .ork file:
*(drag-and-drop the file here as a .zip file)*

### OpenRocket version

unstable

### What platform are you running on?

Linux

### Additional context

Apologies if I missed seeing an existing bug for this same issue.  Also apologies if I've missed another change in the build instructions.

JoePfeiffer, 2025-07-15:
I'm having a hard time duplicating this, running Debian Trixie (unfortunately I don't have any machines running Bullseye or Bookworm). I created a new local repository and built, using:
  684  git clone https://github.com/openrocket/openrocket.git
  685  cd openrocket/
  686  git submodule init
  687  git submodule update --remote
  688  git fetch
  689  git pull
  690  ./gradlew clean submoduleUpdate build dist
  691  ./gradlew run
 (of course the git fetch and git pull didn't do anything in a brand new repository, but I wanted to come as close as possible to duplicating your steps) 
And it ran fine. Since it seems to be getting into trouble reading thrustcurves, do you have any local thrustcurves? Since the specific error is a null file iterator, could there be a permissions issue or could the thrustcurve directory not exist?

questionablerocket, 2025-07-15:
Thanks for confirming.  That gave me more to go on in troubleshooting.  I followed your steps exactly on a blank VM and got a successful build.

I did not believe this could be a problem with motor files, since I encountered the problem on a blank VM, but in reviewing my command history, I see that the first time I tried building on that VM with gradlew, I ran:

./gradlew clean submoduleUpdate serializeEnginesDist build dist

The build in that case failed.  Evidently at the time (it was late at night) I failed to note that down and simply tried rerunning without serializeEnginesDist.  I subsequently ran:

./gradlew clean submoduleUpdate build dist

at which point the build succeeded but failed to launch as previously described.

I have now confirmed this behavior on thee machines, Debian bookworm, Debian bullseye (VM), and MacOS.

I had been in the habit of periodically running serializeEnginesDist in my builds to pick up new motors, and this used to work and pull in the motor files from Thrustcurve.  Now it seems it fails and corrupts the local repository in such a way that subsequent builds hand when trying to load motors.

Blowing away the local repository and running without serializeEnginesDist works.

./gradlew serializeEnginesDist fails with the following message:

     Task :core:serializeEnginesExecuteDist FAILED
     Starting serializeEnginesExecuteDist...
     Usage: <inputDir> <outputDir> <threadUtilFraction>

serializeEnginesDist appears to call serializeEnginesExecuteDist, and I get the same error when running serializeEnginesExecuteDist instead.

I would guess the reason the resultant jar hangs at loading motors is because serializeEnginesExecuteDist triggers serializeEnginesDelete, and after serializeEnginesExecuteDist fails, there are no motors to load.  Both ./gradlew serializeEnginesDelete and ./gradlew serializeEnginesExecute succeed, but both ./gradlew serializeEnginesExecuteDist and ./gradlew serializeEnginesDist fail.

I would also _guess_ this is happening because of the changes core/build.gradle (where serializeEnginesDist appears) and other files as part of the revamp of SerializeThurstcurveMotors that was committed on May 7th.  Unfortunately I don't have the time - and likely not the skills either - to figure out quite where the failure is being introduced, so while I'll close this bug (as I have working builds now), I won't be able to open a coherent bug for serializeEnginesDist/serializeEnginesExecuteDist now or anytime soon.

Thanks again for the sanity check and proving the basic build process worked!

closed 2025-07-15 by questionablerocket

questionablerocket, 2025-07-16:
I've been using serializeEnginesDist because I wasn't sure what to make of the description of the various gradlew commands.  If running serializeEngines brings in the motors and a subsequent "dist" includes them in the build, that's good enough for me.  I will update my build script.

Thanks for the tip!