openrocket/openrocket #2846: Have AppImage pipeline. How to integrate?

eduncan911, 2025-07-24:
I have created a Docker build environment that produces an AppImage that functions on a bare metal archlinux machine.  The only requirement is docker to be installed.  It should work under Podman as well, though I have not tried.  It is based on the snap file (and doesn't work for release-24.12.RC.01, doh).

- completely self contained Eclipse Temurin from Adoptium opensdk build environment within the docker container.
- 23.09 only supported Temurin 17.  However, Temurin 21 can easily be switched to and used for newer branches.
- caching layers are used for faster rebuilds/branch switching (unit testing, etc).

I am unsure about how to start a PR that would integrate this nicely with openrocket's infrastructure, specifically around file placements and naming conventions.  

Overall, I would highly suggest a GitHub Action to build the appimage for every Release published, and automatically attaches.  I have many examples I've done for other orgs here on github.  Make a release, click Publish, and in 5-10 minutes an appimage attaches itself to that release.

The file structure I have currently is:

    <repo>/docker
    <repo>/docker/Dockerfile
    <repo>/docker/openrocket.bash
    <repo>/docker/openrocket.desktop
    <repo>/docker/openrocket.run
    <repo>/docker/Makefile

It doesn't have to be a `Makefile`.  It's just a few lines, can be shell or bash, etc.  It's invoked such as:

    make
    output/OpenRocket-23.09-x86_64.AppImage

Supports branches with a really fast turn around of a few minutes.

    # note: this fails to build with "build.xml missing"
    make BRANCH=release-24.12.RC.01

There's a `run.sh` in the root of the repository.  Perhaps I can add a `build.sh`? 

Thanks!

eduncan911, 2025-07-24:
[image]

SiboVG, 2025-07-24:
_(Not knowing too much about AppImages)_ What is the difference between this and the .sh file we provide with our releases?

eduncan911, 2025-07-25:
> _(Not knowing too much about AppImages)_ What is the difference between this and the .sh file we provide with our releases?

Several significant things.

- the most significant difference is the "shell script" installer requires you have Java installed. In contrast, AppImages do not as they are completely self-contained (all dependencies are packed into a portable file).
- appimages are cross platform (Linux and BSD flavors)
- "snaps" are only for the Ubuntu edition of Linux (large usebase, but a very narrow subset of OS), and generally avoided in the security space for various reasons.  Typically you publish flatpaks, snaps, and an appimage, to cover all Linux and bsd users.  

And general security best practices is never to run arbitrary scripts on your local machine.  Unlike Windows where you grant full permissions to your system for every installer, on Linux we are highly protective of our userpsace, and even more so of the root system.  A 140MB shell script screams embedded binaries as red flags.

eduncan911, 2025-07-25:
> I know a little bit about AppImages (I built one for 15.03 so users wouldn't have to keep an antique version of Java on their systems), but virtually nothing about containers.
> 
> OR as an AppImage would basically be our installed tree, compiled up into one big executable file. A user could just put the file anywhere they want, make it executable, and run it without needing to go through our installer.
> 
> A container, as I understand it, does more sandboxing than a standard application; it has its own view of the filesystem and other system resources other than the kernel itself. The goal is to have more isolation between applications, like a VM, but without the need to boot a kernel.
> 

Thanks for this!! 

- Docker is only used as an isolated, secure build system of where the host machine doesn't need any build tools nor java installed.  it's all self contained within the container and cannot escape.
- Docker's manifesto is to provide a repeatable build system across all platforms, devices, and machines.  In this case, I'm proposing making GitHub Actions be the worker here to further isolate the build and signing process away from developers.

> Your error on 24.12.RC.01 sounds like you're using the ant build system -- build.xml is the configuration file for that.
> 

I know nothing of building Java apps these days.  Is the snap updated for 24 RC?  If so, I'll pull that version down and change the Dockerfile over to this build.xml.  

If snap hasn't been updated, could someone point me to the compilation instructions for this "build.xml" java process?

> My first impulse would be that your docker-related files would go in a docker/ directory, just as the snap related files go in the snap/ directory. I don't honestly remember how snap builds are triggered... 
> 

Well,. docker is only a build tool here, not a runtime.  What about `/appimage` instead following the logic of /snap? 

> one question that comes to mind is whether, instead of introducing AppImages, the docker image could be built from the snap (I ask this knowing even less about snap than I do docker).
> 

Snaps are basically proprietary only to the Ubuntu flavor of Linux - this significantly limits openrocket's distribution.  They have the same basic build structure and sandboxing runtimes.  But instead of using the snap build tools, I use docker as my universal build environment for my build tools, that runs on all flavors of Linux and BSD.  Then, I produce an AppImage making the runtime universal across all Linux machines.

eduncan911, 2025-07-25:
With AppImage, Flatpak, and Snap, we can drop the shell script runtime from distribution all together.

JoePfeiffer, 2025-07-25:
A few clarifications --

The installer does not require the user to have a java environment on their machine. It includes its own JRE, which it unpacks at the start of the process.

An AppImage is less cross-platform than OR -- we aren't just Linux and BSD, we are also MacOS and Windows. We will need to continue to have our current installer for those environments. The installer builder is also not only cross platform but cross architecture -- we're able to build both x86-64 and ARM architectures with it.

I wasn't clear enough about build.xml -- we have switched to the gradle build system so there is no build.xml any more.

We really want the build process to be as close to the same for all of the environments and architectures we support. At present it's virtually identical for all of them, and I can build for all of them from my Linux workstation. Our Windows and MacOS based developers can do the same -- we really don't want to lose that.

I regard our snap experience as something of a cautionary tale: the developer who did that has left the project, the switch to gradle for the build system appears not to have worked, so any users who are using snaps aren't getting updates.

JoePfeiffer, 2025-07-27:
The installer script does not install java on the system. The only java present after running the installer script is inside the OpenRocket directory structure. The AppImage has to also contain a JRE, else the .jar file cannot be executed.

Let's open a PR putting it in a top-level directory named appimage.