openrocket/openrocket #2855: TEKNOFEST Rocket Competition and AeroTech engine incompatibility

mehmeterdengecti, 2025-07-28:
Teknofest gives its own dimensions for the AeroTech rocket engines used in the TEKNOFEST rocket competition( the rocket engine dimensions given by Teknofest itself.)fest gives its own dimensions for the AeroTech rocket engines used in the TEKNOFEST rocket competition( the rocket engine dimensions given by Teknofest itself.) The dTeknofest gives its own dimensions for the AeroTech rocket engines used in the TEKNOFEST rocket competition( the rocket engine dimensions given by Teknofest itself.) The dimensions provided by Teknofest do not match the worry.
@tibbe @neilweinstock @soupwizard @itshanney @UncleRus  

[image]

[image]

[image]

 There is a video of the error on the drive link : [https://drive.google.com/file/d/13mhumyKaDXdOJVwadV0nEN6qdDHpzUjU/view?usp=sharing](url)

ProtonNumber, 2025-07-28:
You shouldn't be putting the .eng file in "user defined component presets".

Click reset on that and try it again?

mehmeterdengecti, 2025-07-28:
> Raporunuz için teşekkürler. Lütfen teknofest_l1256.eng dosyasını buraya yükleyebilir misiniz?
Hey friend, Github does not allow me to upload files with .eng extension. I'm leaving the Drive link:  
https://drive.google.com/drive/folders/1YYre34S3No6xlKA23t_vPwQj4m85WziE?usp=sharing

mehmeterdengecti, 2025-07-28:
I accidentally closed the discussion, how do I reopen it?

mehmeterdengecti, 2025-07-28:
> You shouldn't be putting the .eng file in "user defined component presets".
> 
> Click reset on that and try it again?

Can you explain a little bit how I can do what you call a friend?

SiboVG, 2025-07-28:
@ProtonNumber is right. You have added your thrust curve (.eng file) in the component presets instead of in the thrust curve section:

[image]

1. Delete the .eng file from the "User-defined component preset" text field, either by manually deleting the file path text, or by clicking the "Reset' button to the right of it
2. Add the .eng file in the "User-defined thrust curves" section

In any case, we will try to make this section of the program more fool-proof, so thank you for reporting it!

mehmeterdengecti, 2025-07-28:
I prepared the text of my report using Yandex Translate, but it was really poorly prepared. If there is anything you don't understand, feel free to ask.

closed 2025-07-28 by mehmeterdengecti