openrocket/openrocket #2871: OpenRocket Bug

mafefaiq-sudo, 2025-08-09:
---------- Bug report ----------

Please include a description about what actions you were performing when the exception occurred:
(You can edit text directly in this window)

1. Unable to edit nose cone
2. Crash while editing (PC still working)
3. Cannot fix the bug

If possible, please send us the .ork file that caused the bug.

Include your email address (optional; it helps if we can contact you in case we need additional information):

(Do not modify anything below this line.)
---------- Exception stack trace ----------
java.lang.NullPointerException: Cannot invoke "info.openrocket.core.preset.xml.OpenRocketComponentDTO.asComponentPresets()" because the return value of "info.openrocket.core.preset.xml.OpenRocketComponentSaver.unmarshalFromOpenRocketComponent(java.io.Reader)" is null at info.openrocket.core.preset.xml.OpenRocketComponentLoader.load(OpenRocketComponentLoader.java:35) at info.openrocket.core.database.ComponentPresetDatabaseLoader.loadFile(ComponentPresetDatabaseLoader.java:117) at info.openrocket.core.database.ComponentPresetDatabaseLoader.loadUserComponents(ComponentPresetDatabaseLoader.java:73) at info.openrocket.core.database.ComponentPresetDatabaseLoader.loadDatabase(ComponentPresetDatabaseLoader.java:55) at info.openrocket.core.database.AsynchronousDatabaseLoader.doLoad(AsynchronousDatabaseLoader.java:108) at info.openrocket.core.database.AsynchronousDatabaseLoader$LoadingThread.run(AsynchronousDatabaseLoader.java:144)

---------- Thread information ----------
Thread[DatabaseLoadingThread,1,main]

---------- System information ----------
OpenRocket version: 24.12
OpenRocket source: default
OpenRocket location: C:\Users\Zulfakri\Desktop\UmarWafi\MRC\OpenRocket\.\jar\OpenRocket-24.12.jar
User-defined thrust curves location: C:\Users\Zulfakri\AppData\Roaming\OpenRocket\ThrustCurves;C:\Users\Zulfakri\Documents\R1KM-KNSB_ClassI_v4 (2).eng;C:\Users\Zulfakri\Documents\R1KM-KNSB_ClassI_v4 (2).eng
LAF: com.formdev.flatlaf.FlatLightLaf
JOGL version: null
Current default locale: en_GB
JVM Arguments: -Dexe4j.semaphoreName=Local\c:_users_zulfakri_desktop_umarwafi_mrc_openrocket_openrocket.exe1 -Dexe4j.isInstall4j=true -Dexe4j.moduleName=C:\Users\Zulfakri\Desktop\UmarWafi\MRC\OpenRocket\OpenRocket.exe -Dexe4j.tempDir= -Dexe4j.unextractedPosition=0 -Djava.library.path=c:\users\zulfakri\desktop\umarwafi\mrc\openrocket\jre\bin;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files\Git\cmd;C:\Program Files\nodejs\;C:\Program Files\PuTTY\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin\;C:\Program Files\IVI Foundation\VISA\Win64\Bin\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\dotnet\;C:\Program Files (x86)\Windows Kits\10\Windows Performance Toolkit\;C:\Users\Zulfakri\AppData\Local\Programs\Python\Python312\Scripts\;C:\Users\Zulfakri\AppData\Local\Programs\Python\Python312\;C:\Users\Zulfakri\AppData\Local\Programs\Python\Launcher\;C:\Users\Zulfakri\AppData\Local\Programs\Python\Python310\Scripts\;C:\Users\Zulfakri\AppData\Local\Programs\Python\Python310\;C:\Users\Zulfakri\AppData\Local\Microsoft\WindowsApps;C:\Users\Zulfakri\AppData\Roaming\npm;C:\Users\Zulfakri\AppData\Local\Programs\Microsoft VS Code\bin;C:\php;C:\Users\Zulfakri\AppData\Local\atom\bin;C:\Program Files\JetBrains\PyCharm Community Edition 2022.2.2\bin;C:\ProgramData\Zulfakri\atom\bin -Dexe4j.consoleCodepage=cp0 -Dapple.awt.application.appearance=system -Dsun.java2d.noddraw=true -Dsun.java2d.d3d=false -Dsun.java2d.ddforcevram=true -Dsun.java2d.ddblit=false -Dswing.useflipBufferStrategy=True --add-exports=java.base/java.lang=ALL-UNNAMED --add-exports=java.desktop/sun.awt=ALL-UNNAMED --add-exports=java.desktop/sun.java2d=ALL-UNNAMED -Dinstall4j.launcherId=59 -Dinstall4j.swt=false
System properties:
apple.awt.application.appearance=system
exe4j.consoleCodepage=cp0
exe4j.isInstall4j=true
exe4j.launchName=C:\Users\Zulfakri\Desktop\UmarWafi\MRC\OpenRocket\OpenRocket.exe
exe4j.moduleName=C:\Users\Zulfakri\Desktop\UmarWafi\MRC\OpenRocket\OpenRocket.exe
exe4j.semaphoreName=Local\c:_users_zulfakri_desktop_umarwafi_mrc_openrocket_openrocket.exe1
exe4j.tempDir=
exe4j.unextractedPosition=0
file.encoding=MS932
file.separator=\
flatlaf.uiScale=1.0
install4j.appDir=C:\Users\Zulfakri\Desktop\UmarWafi\MRC\OpenRocket\
install4j.exeDir=C:\Users\Zulfakri\Desktop\UmarWafi\MRC\OpenRocket\
install4j.launcherId=59
install4j.swt=false
java.class.path=C:\Users\Zulfakri\Desktop\UmarWafi\MRC\OpenRocket\.install4j\i4jruntime.jar;C:\Users\Zulfakri\Desktop\UmarWafi\MRC\OpenRocket\.\jar\OpenRocket-24.12.jar
java.class.version=61.0
java.home=c:\users\zulfakri\desktop\umarwafi\mrc\openrocket\jre
java.io.tmpdir=C:\Users\Zulfakri\AppData\Local\Temp\
java.library.path=c:\users\zulfakri\desktop\umarwafi\mrc\openrocket\jre\bin;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files\Git\cmd;C:\Program Files\nodejs\;C:\Program Files\PuTTY\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin\;C:\Program Files\IVI Foundation\VISA\Win64\Bin\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\dotnet\;C:\Program Files (x86)\Windows Kits\10\Windows Performance Toolkit\;C:\Users\Zulfakri\AppData\Local\Programs\Python\Python312\Scripts\;C:\Users\Zulfakri\AppData\Local\Programs\Python\Python312\;C:\Users\Zulfakri\AppData\Local\Programs\Python\Launcher\;C:\Users\Zulfakri\AppData\Local\Programs\Python\Python310\Scripts\;C:\Users\Zulfakri\AppData\Local\Programs\Python\Python310\;C:\Users\Zulfakri\AppData\Local\Microsoft\WindowsApps;C:\Users\Zulfakri\AppData\Roaming\npm;C:\Users\Zulfakri\AppData\Local\Programs\Microsoft VS Code\bin;C:\php;C:\Users\Zulfakri\AppData\Local\atom\bin;C:\Program Files\JetBrains\PyCharm Community Edition 2022.2.2\bin;C:\ProgramData\Zulfakri\atom\bin
java.runtime.name=OpenJDK Runtime Environment
java.runtime.version=17.0.16+12-LTS
java.specification.maintenance.version=1
java.specification.name=Java Platform API Specification
java.specification.vendor=Oracle Corporation
java.specification.version=17
java.util.Arrays.useLegacyMergeSort=true
java.vendor=BellSoft
java.vendor.url=https://bell-sw.com/
java.vendor.url.bug=https://bell-sw.com/support
java.version=17.0.16
java.version.date=2025-07-15
java.vm.compressedOopsMode=32-bit
java.vm.info=mixed mode
java.vm.name=OpenJDK 64-Bit Server VM
java.vm.specification.name=Java Virtual Machine Specification
java.vm.specification.vendor=Oracle Corporation
java.vm.specification.version=17
java.vm.vendor=BellSoft
java.vm.version=17.0.16+12-LTS
[... 158 lines cut ...]
229 23.839 INFO (SwingExceptionHandler.java:65) Exception handler not on EDT, invoking dialog on EDT
230 24.364 DEBUG (GUIUtil.java:369) Storing size of info.openrocket.swing.gui.main.BasicFrame: java.awt.Dimension[width=1382,height=784]
231 24.364 DEBUG (GUIUtil.java:374) Storing maximized state of info.openrocket.swing.gui.main.BasicFrame
232 24.364 DEBUG (GUIUtil.java:369) Storing size of info.openrocket.swing.gui.main.BasicFrame: java.awt.Dimension[width=1382,height=784]
233 24.364 DEBUG (GUIUtil.java:374) Storing maximized state of info.openrocket.swing.gui.main.BasicFrame
234 24.374 DEBUG (GUIUtil.java:369) Storing size of info.openrocket.swing.gui.main.BasicFrame: java.awt.Dimension[width=1382,height=784]
235 24.374 DEBUG (GUIUtil.java:374) Storing maximized state of info.openrocket.swing.gui.main.BasicFrame
236 24.374 DEBUG (GUIUtil.java:369) Storing size of info.openrocket.swing.gui.main.BasicFrame: java.awt.Dimension[width=1382,height=784]
237 24.374 DEBUG (GUIUtil.java:374) Storing maximized state of info.openrocket.swing.gui.main.BasicFrame
238 24.374 VBOSE (AbstractScaleFigure.java:209) ____ Updating RocketFigure to: 0.130813 user scale, 494.411 overall scale
239 24.374 VBOSE (AbstractScaleFigure.java:209) ____ Updating RocketFigure to: 0.130813 user scale, 494.411 overall scale
240 24.384 VBOSE (AbstractScaleFigure.java:209) ____ Updating RocketFigure to: 0.130813 user scale, 494.411 overall scale
241 24.631 DEBUG (GUIUtil.java:369) Storing size of info.openrocket.swing.gui.main.BasicFrame: java.awt.Dimension[width=1382,height=784]
242 24.631 DEBUG (GUIUtil.java:374) Storing maximized state of info.openrocket.swing.gui.main.BasicFrame
243 24.641 VBOSE (AbstractScaleFigure.java:209) ____ Updating RocketFigure to: 0.130813 user scale, 494.411 overall scale
[... cut ...]
248 24.926 INFO (SwingExceptionHandler.java:202) Showing Exception dialog
249 24.996 VBOSE (AbstractScaleFigure.java:209) ____ Updating RocketFigure to: 0.130813 user scale, 494.411 overall scale
250 24.996 VBOSE (AbstractScaleFigure.java:209) ____ Updating RocketFigure to: 0.130813 user scale, 494.411 overall scale
251 26.629 USER (SwingExceptionHandler.java:221) User requested sending bug report
252 26.639 DEBUG (BugReportDialog.java:230) Found jar file using codeSource
---------- End of bug report ----------

JoePfeiffer, 2025-08-09:
I'm having a hard time replicating your bug. Could you give us more detail with a specific set of steps that will trigger it? For instance
1. Run OR
2. Open attached .ork file
3. Select nose cone
4. Click "edit"
and so forth.

Thanks!

SiboVG, 2025-09-13:
Could you please tell us the content of your user-defined component presets field in the preferences dialog?

closed 2025-11-20 by SiboVG