openrocket/openrocket #2873: Error during save of project

DDani47, 2025-08-17:
I was going to save a project, when an error message appeared, and i don't know why this happens 

[

[image]

](url)

SiboVG, 2025-08-17:
Please click "View bug report" and paste the text from the bug report dialog here.

DDani47, 2025-08-17:
---------- Exception stack trace ----------
java.lang.NullPointerException: Cannot invoke "info.openrocket.core.rocketcomponent.RocketComponent.getID()" because "c" is null at info.openrocket.core.file.openrocket.OpenRocketSaver.saveSimulation(OpenRocketSaver.java:449) at info.openrocket.core.file.openrocket.OpenRocketSaver.save(OpenRocketSaver.java:112) at info.openrocket.core.file.GeneralRocketSaver.saveInternal(GeneralRocketSaver.java:267) at info.openrocket.core.file.GeneralRocketSaver.saveAllPartsZipFile(GeneralRocketSaver.java:227) at info.openrocket.core.file.GeneralRocketSaver.save(GeneralRocketSaver.java:213) at info.openrocket.core.file.GeneralRocketSaver.save(GeneralRocketSaver.java:122) at info.openrocket.core.file.GeneralRocketSaver.save(GeneralRocketSaver.java:92) at info.openrocket.swing.gui.util.SaveFileWorker.doInBackground(SaveFileWorker.java:37) at info.openrocket.swing.gui.util.SaveFileWorker.doInBackground(SaveFileWorker.java:11) at java.desktop/javax.swing.SwingWorker$1.call(SwingWorker.java:304) at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264) at java.desktop/javax.swing.SwingWorker.run(SwingWorker.java:343) at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136) at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635) at java.base/java.lang.Thread.run(Thread.java:840)

---------- Thread information ----------
Thread[AWT-EventQueue-0,6,main]

---------- System information ----------
OpenRocket version: 24.12
OpenRocket source: default
OpenRocket location: C:\Program Files\OpenRocket\.\jar\OpenRocket-24.12.jar
User-defined thrust curves location: C:\Users\LENOVO M720S\AppData\Roaming\OpenRocket\ThrustCurves
LAF: com.formdev.flatlaf.FlatLightLaf
JOGL version: null
Current default locale: es_CO
JVM Arguments: -Dexe4j.semaphoreName=Local\c:_program_files_openrocket_openrocket.exe0 -Dexe4j.isInstall4j=true -Dexe4j.moduleName=C:\Program Files\OpenRocket\OpenRocket.exe -Dexe4j.tempDir= -Dexe4j.unextractedPosition=0 -Djava.library.path=c:\program files\openrocket\jre\bin;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\dotnet\;C:\Users\LENOVO M720S\AppData\Local\Microsoft\WindowsApps;C:\Users\LENOVO M720S\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\LENOVO M720S\AppData\Local\Muse Hub\lib -Dexe4j.consoleCodepage=cp0 -Dapple.awt.application.appearance=system -Dsun.java2d.noddraw=true -Dsun.java2d.d3d=false -Dsun.java2d.ddforcevram=true -Dsun.java2d.ddblit=false -Dswing.useflipBufferStrategy=True --add-exports=java.base/java.lang=ALL-UNNAMED --add-exports=java.desktop/sun.awt=ALL-UNNAMED --add-exports=java.desktop/sun.java2d=ALL-UNNAMED -Dinstall4j.launcherId=59 -Dinstall4j.swt=false
System properties:
apple.awt.application.appearance=system
exe4j.consoleCodepage=cp0
exe4j.isInstall4j=true
exe4j.launchName=C:\Program Files\OpenRocket\OpenRocket.exe
exe4j.moduleName=C:\Program Files\OpenRocket\OpenRocket.exe
exe4j.semaphoreName=Local\c:_program_files_openrocket_openrocket.exe0
exe4j.tempDir=
exe4j.unextractedPosition=0
file.encoding=Cp1252
file.separator=\
flatlaf.uiScale=1.0
install4j.appDir=C:\Program Files\OpenRocket\
install4j.exeDir=C:\Program Files\OpenRocket\
install4j.launcherId=59
install4j.swt=false
java.class.path=C:\Program Files\OpenRocket\.install4j\i4jruntime.jar;C:\Program Files\OpenRocket\.\jar\OpenRocket-24.12.jar
java.class.version=61.0
java.home=c:\program files\openrocket\jre
java.io.tmpdir=C:\Users\LENOVO~1\AppData\Local\Temp\
java.library.path=c:\program files\openrocket\jre\bin;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\dotnet\;C:\Users\LENOVO M720S\AppData\Local\Microsoft\WindowsApps;C:\Users\LENOVO M720S\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\LENOVO M720S\AppData\Local\Muse Hub\lib
java.runtime.name=OpenJDK Runtime Environment
java.runtime.version=17.0.16+12-LTS
java.specification.maintenance.version=1
java.specification.name=Java Platform API Specification
java.specification.vendor=Oracle Corporation
java.specification.version=17
java.util.Arrays.useLegacyMergeSort=true
java.vendor=BellSoft
java.vendor.url=https://bell-sw.com/
java.vendor.url.bug=https://bell-sw.com/support
java.version=17.0.16
java.version.date=2025-07-15
java.vm.compressedOopsMode=Zero based
java.vm.info=mixed mode
java.vm.name=OpenJDK 64-Bit Server VM
java.vm.specification.name=Java Virtual Machine Specification
java.vm.specification.vendor=Oracle Corporation
java.vm.specification.version=17
java.vm.vendor=BellSoft
java.vm.version=17.0.16+12-LTS
jdk.debug=release
jnlp.jogamp.tmp.cache.root=jln6893146980008570954
line.separator=\u000d\u000a
native.encoding=Cp1252
os.arch=amd64
os.name=Windows 10
os.version=10.0
path.separator=;
sun.arch.data.model=64
sun.awt.enableExtraMouseButtons=true
sun.awt.exception.handler=info.openrocket.swing.gui.main.SwingExceptionHandler$AwtHandler
sun.boot.library.path=c:\program files\openrocket\jre\bin
sun.cpu.endian=little
sun.cpu.isalist=amd64
[... 210 lines cut ...]
984 28.001 DEBUG (SaveFileWorker.java:37) Saving component Aletas trapezoidales
985 28.091 INFO (SwingWorkerDialog.java:151) Worker completed before opening dialog
[... cut ...]
1001 28.091 STDERR (SwingExceptionHandler.java:50) at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635)
1002 28.091 STDERR (SwingExceptionHandler.java:50) at java.base/java.lang.Thread.run(Thread.java:840)
1003 28.091 INFO (SwingExceptionHandler.java:62) Exception handler running on EDT, showing dialog
1004 28.091 INFO (SwingExceptionHandler.java:202) Showing Exception dialog
1005 29.867 USER (SwingExceptionHandler.java:221) User requested sending bug report
1006 29.867 DEBUG (BugReportDialog.java:230) Found jar file using codeSource
---------- End of bug report ----------

JoePfeiffer, 2025-08-17:
Is this repeatable (ie if you rerun the sim and try to save again, do you get the error again)? It looks like your simulation probably had a warning -- what warning or warnings did you have?

closed 2025-12-06 by SiboVG