openrocket/openrocket #2885: [Feature Request] Unzipped .xml support

yevagorbachev, 2025-09-06:
**Is your feature request related to a problem? Please describe.**
OR models are not friendly to version control systems.

**Describe the solution you'd like**
Open .xml files directly

**Describe alternatives you've considered**
Team-built CI that somehow automatically synchronizes an .xml with an .ork

yevagorbachev, 2025-09-06:
Specifically, I like that .xml files have informative diffs as plain text. While embedded binary files like images require the directory structure a .zip allows, supporting just the rocket and sim definition will be a huge step up.

yevagorbachev, 2025-09-06:
Compatibility with a feature like #2884 is preserved by embedding the thrust data into the .xml file. If I understand correctly, that's how multi level winds work already.

SiboVG, 2025-09-06:
This is not possible if the design uses decals. Decals are saved together with the Rocket.xml, hence why OpenRocket is a zipped file.

yevagorbachev, 2025-09-06:
Sure. The point I made in my first comment on this issue is that it would be nice if OpenRocket supported opening decal-less rockets as .xml files, because the .xml file is what contains everything that actually matters (ideally, see #1838) 

I am not sure there is much overlap between decal lovers and people who care about VCS diffs anyway...

JoePfeiffer, 2026-08-27:
> Compatibility with a feature like [#2884](https://github.com/openrocket/openrocket/issues/2884) is preserved by embedding the thrust data into the .xml file. If I understand correctly, that's how multi level winds work already.

Putting thrustcurves in different files, rather than in the .ork, is a deliberate design choice. It's a bit of a niche interest, but it makes it possible to share EX motors easily.

kjhambrick, 2026-08-27:
> Putting thrustcurves in different files, rather than in the .ork, is a deliberate design choice. It's a bit of a niche interest, but it makes it possible to share EX motors easily.

Thanks for this new SNAPSHOT 26XX feature Joe and OR devs !

The new explicit thrustcurves/ folder in the .ORK will solve an old issue where I have a few personal EMK motors in my .ork files.

In 2412 and before, there would be errors whenever I shared such an ORK with someone else unless they installed my thrust curves on their system before opening the .ork ...

My $0.02 worth:  Nice one !

-- kjh

p.s. and now that I know how to properly unzip an ORK, all my old ORK processing scripts will work, even the ones with decals :)

kjhambrick, 2026-08-27:
> Sure. The point I made in my first comment on this issue is that it would be nice if OpenRocket supported opening decal-less rockets as .xml files, because the .xml file is what contains everything [...]

Hmmm ...

Would your VCS workflow support a pre-processing script to unzip the OR.ork into a named directory then run your VCS on the rocket.ork file in the named directory ?

Then a post-processing script after a checkout to rebuild an OR.ork with all the original structure from within the named directory ?

Wild idea but HTH ...

-- kjh

yevagorbachev, 2026-08-27:
> Putting thrustcurves in different files, rather than in the .ork, is a deliberate design choice. It's a bit of a niche interest, but it makes it possible to share EX motors easily.

I don't see why they should be mutually exclusive. 

It seems like this is a choice that makes it easier to share motors than to share rockets, which feels backwards; I'd guess that rockets are shared far more often than motors.

EDIT: I missed the detail that this is a "feature" now, and you were still referring to embedding, but just as a separate file instead of XML tag. Sorry!

yevagorbachev, 2026-08-27:
> Hmmm ...
>
> Would your VCS workflow support a pre-processing script to unzip the OR.ork into a named directory then run your VCS on the rocket.ork file in the named directory ?
>
> Then a post-processing [...]

This is theoretically doable and objectively not hard, but at the time it seemed to me that it would be easier to make ORK open .xml files (and solves the problem at one point, otherwise every team is out here rolling their own git hooks). If there are no decals; fine: I care that a sim on one computer runs and produces the same trajectory (up to the random seeding) as on another.