openrocket/openrocket #2893: Cannot export plot

BhavyaFCHS, 2025-09-19:
---------- Bug report ----------

[image]

[image]

Please include a description about what actions you were performing when the exception occurred:
(You can edit text directly in this window)

1. The plot shows up just fine, but when exporting the y-axis variable with the same component selected it gives this error.
2. Field separator string is ","
3. No comments

If possible, please send us the .ork file that caused the bug.

Include your email address (optional; it helps if we can contact you in case we need additional information):

(Do not modify anything below this line.)
---------- Exception stack trace ----------
java.lang.NullPointerException: Cannot invoke "java.util.List.isEmpty()" because the return value of "java.util.Map.get(Object)" is null at info.openrocket.swing.gui.dialogs.componentanalysis.CAExportPanel.doExport(CAExportPanel.java:150) at info.openrocket.swing.gui.dialogs.componentanalysis.ComponentAnalysisPlotExportPanel$3.actionPerformed(ComponentAnalysisPlotExportPanel.java:217)

SiboVG, 2025-09-20:
Are you running the latest version, version 24.12 or 24.12.RC.01? I recall this bug (or a similar one) occurring before the 24.12 release.

SiboVG, 2025-10-19:
No response from user.

closed 2025-10-19 by SiboVG